arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #2: Jpg page load finishes late

s0170071, 2020-09-16:
When loading the JPG, the image appears instantly in the browser but the network connection remains open for about 2 seconds.

[image]

arkhipenko, 2020-09-16:
I am not doing anything special with the network connection as you can see from the code. Not sure what you are asking me to do about this?

arkhipenko, 2020-09-16:
I was not paying much attention to the JPG aspect of it - the focus was on the video feed obviously.
Would adding `client.stop()` after
https://github.com/arkhipenko/esp32-mjpeg-multiclient-espcam-drivers/blob/master/esp32-cam/esp32-cam.ino#L368
help?

s0170071, 2020-09-16:
The issue is that the image is already displayed in the browser but it takes another two seconds until it stops loading. It seems the browser is expecting something, i.e. a \r\n.  
I already tried Client.stop but that didn't help. The streaming (also jpg) is done asynchronously, that may be why.

The long loading times cause issues with my home automation. It times out.