arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #3: Feature request: PIO + EmbUI + ESPAsyncWebServer support/example

DmytroKorniienko, 2020-10-28:
Hi. How are you? Hope you doing well.

I started new fork which based on your repo and want to mix out EmbUI framework with esp32-mjpeg-multiclient code. I prefered working in the VS-Code + PIO, so had migrated your code into it. In addition EmbUI using ESPAsyncWebServer instead Webserver, so I changed some calls as well. Now all partially worked, but unfortuanetelly I have broking output only small part of buffer. Can you look at code or provide an example for ESPAsyncWebServer?

And many thanks for your work!

[image]

arkhipenko, 2020-10-28:
Which of the 3 flavors are you using?

DmytroKorniienko, 2020-10-28:
> Which of the 3 flavors are you using?
\esp32-mjpeg-multiclient-espcam-drivers\version\esp32-cam-rtos-allframes\esp32-cam-rtos-allframes
* issue only for ESPAsyncWebServer, ordinary web server worked as expected

DmytroKorniienko, 2020-10-29:
Nevermind. I found an example: https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3 :)

closed 2020-10-29 by DmytroKorniienko