arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #4: Feedback: Static IP, Watchdog

baden03, 2020-11-22:
First, I must say that after going through a journey of flashing 5 different esp32 cam projects I _finally_ landed on this wonderful repo. It works exactly as hoped, and so far it seems very stable...
...HUZZAH! (and thank you, multiclient (up to 10) times)

The ability to add a static IP is missing, but I know I can easily [implemented](https://randomnerdtutorials.com/esp32-cam-static-fixed-ip-address-arduino/) this feature. Pull request? Stay tuned.

Finally, implementing a [watchdog](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html) in case the esp32 becomes unresponsive would be ideal, as this kind of project lends itself to hard to reach mounting locations.  Pull request? Hmm.

All in all, this project is ACES in my book. Again, thank you.

arkhipenko, 2020-11-22:
@baden03 Thank you for kind words.

As far as static IP is concerned I prefer to assign permanent lease on the router, that way you have more control (IMHO).

As far as watchdog, I would not worry about it - esp32 already have watchdogs on every core. If it becomes unresponsive, it will reset by itself.  Sometimes it is more trouble to actually satisfy the existing watchdogs, let alone create new one. 
However, if you want a special purpose watchdog, like the one to make sure frames are captured at a certain interval, then yes, be my guest!

This is not a final product, more of a POC and a framework for others.

Enjoy.

baden03, 2020-11-24:
I was not aware of the core-watchdog, thanks for the info.
Now that I have had it installed and running non-stop for 24 hours, I am even more impressed with the project!

The permanent lease on the router is a workable solution, but a option for a static IP might be useful. I'll add it to my 'things to do while in lock-down' list :)

arkhipenko, 2021-03-08:
Feel free to implement changes and/or submit a PR.

closed 2021-03-08 by arkhipenko