arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #6: Cant connect to Wifi

slipx06, 2020-12-07:
Tried both the esp32-cam and esp32-cam-allframes sketches but in both instances I'm unable to connect to the WIFI network. The following output is from the serial monitor using an AI thinker board. Any ideas?

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
E (5536) wifi: Expected to init 16 rx buffer, actual is 9
E (5554) wifi: Expected to init 16 rx buffer, actual is 0
Connecting to WiFi..........................................................................................

arkhipenko, 2020-12-07:
I assume you replaced this with your credentials:

```
/*
  Next one is an include with wifi credentials.
  This is what you need to do:
  1. Create a file called "home_wifi_multi.h" in the same folder   OR   under a separate subfolder of the "libraries" folder of Arduino IDE. (You are creating a "fake" library really - I called it "MySettings").
  2. Place the following text in the file:
  #define SSID1 "replace with your wifi ssid"
  #define PWD1 "replace your wifi password"
  3. Save.
  Should work then
*/
#include "home_wifi_multi.h"
```

and the credentials are correct. 
Remember - esp32 cannot connect to 5Ghz networks. 
Are you too far from the AP? Esp32-cam antennas are not very robust.

slipx06, 2020-12-07:
Yes I've updated with the correct credentials and I'm connecting to a 2.4Ghz network. Tried a different sketch and it connected fine so not sure.

arkhipenko, 2020-12-07:
Do you provide ESP32 with enough power?

slipx06, 2020-12-07:
Yes. Using a dedicated 5v 2A power supply

arkhipenko, 2020-12-07:
I have not seen these error before. Try googling it?
```
E (5536) wifi: Expected to init 16 rx buffer, actual is 9
E (5554) wifi: Expected to init 16 rx buffer, actual is 0
```
It obviously works on my end. 
Do you have another esp-cam you can test this on?
I noticed that they are not the same, depending on where you order from. Some work better than others.

slipx06, 2020-12-07:
I ordered 2 boards. Same error on both of them. Will dig a bit more

arkhipenko, 2020-12-07:
Try just a simple sketch with wifi and nothing else. If that works, then we can look at my code. If that also does not work, the problem is elsewhere...

closed 2020-12-07 by arkhipenko

slipx06, 2020-12-07:
Loaded up a simple sketch with wifi only and the same issue.

slipx06, 2020-12-07:
Managed to get your sketch to work. Like you said some boards are better than others