arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #7: Sketches (esp32-cam, rtos and rtos all frames) severe lag

Captainoddvious, 2021-01-05:
There is a huge amount of delay in the web stream through browser, 5-10 seconds and frame rate is very poor. With standard example webcam sketch i can get ~4-8 fps with uxga,  is there something i am missing?

arkhipenko, 2021-01-05:
The only time I had this issue was due to a faulty esp32-can module. Some of them are really poor quality. 
As you can see on the video in this post, esp-eye was streaming with practically no delay at all to multiple browser windows.... 

https://www.hackster.io/anatoli-arkhipenko/multi-client-mjpeg-streaming-from-esp32-47768f

Bazmundi, 2021-03-08:
I have it using two different t-camera v1.6.2 and am seeing quite a bit of lag.  Depends on resolution camera is set to and whether you select allframes (which obviously spools).  Regardless, it looks to me as there is one task only handling all clients.   Often, in threaded environments, a thread per connection is used.  Is there much work in looking at a task per client?  It may or may not help with the lag.  Worth and experiment?

arkhipenko, 2021-03-08:
It already creates a dedicated task per video stream. Please read the code before posting.
I do not have severe lag on my esp-eye and esp32-cam modules, so not sure what is going on in your case. Sorry.

Bazmundi, 2021-03-08:
Cheers. Not fully tested yet. Understood.

arkhipenko, 2021-03-08:
Sorry - did not mean to be rude. I can only fix what I can replicate, and so far I was not able to.
I observe a **little** lag at higher resolution, but still am able to stream at ~10-15 FPS. 
I attribute it to the latency of the PSRAM. 
The only time I had a really severe lag (it wasn't even a lag - it was not streaming at all) was some (probably bootlegged) esp32-cam unit. Never happened since.
The network latency plays a role. If you streaming via the internet it will be slower. The fastest of course is on WiFi/same router.

Bazmundi, 2021-03-08:
I didn't take it as rude, so don't  sweat that. I get you can't test
everything. I will poke about to see what I can discover. You've provided
some hints, thanks.

arkhipenko, 2021-03-08:
Anyway, not sure I am able to help here. 
Particularly I do not own the device you are using and it is not even listed as supported. 
What I observe is this: https://youtu.be/bsAVJSZeSmc
A near-real-time performance on ESP-EYE (and slightly worse on ESP32-CAM)

closed 2021-03-08 by arkhipenko