arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #11: Have you tried multi-client in RTSP?

kenneth558, 2021-04-13:
Seems to me that RTSP would be the right philosophy for RTOS, but mainly because everything else in my surveillance system runs on RTSP.  Never having compared before, I wonder if I can couple your multi-client aspects with stream conversion to RTSP.  Have you ruled that out in some way?

arkhipenko, 2021-04-13:
I found RTSP being too laggy - to the tune of several seconds + long start time. MJPEG is much more responsive and there is almost no start lag. 

You can by all means couple RTSP streaming with the multi-client approach, but I have not done that.

closed 2021-04-13 by arkhipenko

arkhipenko, 2021-04-13:
Good place to start is here:
https://www.hackster.io/BnBe_Club/9-rtsp-video-streamer-using-the-esp32-cam-board-8832b4
you'll have to look into CRtspStreamer files

kenneth558, 2021-04-13:
The lad and delay you mentioned brings me to your side, considering the consequence;  I'll add the new devices using the unconverted way first.  If not enough devices can share the wifi bandwidth, then I'll consider converting.  I'm thinking of four devices per corner of the building plus some out at the corner of the property looking in.  We both know I'll never get them all done, but let's see what the bottleneck will end up being.  I want to keep the ability to save 24x7 video to .mp4 segments (6 mins each) but also allow up to 2 persons off-site to log in and watch in real time like my current cameras can do.  Motion detect may have to get bumped, I don't know; I would offload motion-detect, etc. to [an] SBC (Pine-64), but my main server (3.5TB+) saves the data in the end.

arkhipenko, 2021-04-13:
A better solution for the security system is to have each esp32 cam push data to some backend instead of being a server. I can implement that if you want to hire me on Upwork.

kenneth558, 2021-04-13:
I really do appreciate the offer, but I have an income of $0/year.  I'll give you 25% of it, but I'm sure you can find something more profitable.  I always have to do my own development.  This project is simply for security of my future partial inheritance, so it won't make any money.  Sorry about that

arkhipenko, 2021-04-14:
No worries. Good luck!