arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #12: GitHub doesn't give .zip, it gives .xz

kenneth558, 2021-04-14:
I'm troubleshooting numerous  errors.  After several "multiple definition" errors when compiling, **I deleted and reloaded everything ESP32**, and now I start off with  compile error "no such file..." for esp_err.h.   Then I look and see

#include "esp_err.h"
#include "driver/ledc.h"
#include "sensor.h"
#include "sys/time.h"

Maybe I'm missing something, but I don't see those four files supplied by either you or Espressif.  Did Espressif remove them from this version 1.0.6 Library, I wonder? 

So I'm looking closer at your instructions.

You say "Recompiled with ESP32 **Arduino** Core 1.0.5".  Do you mean "Recompiled with ESP32 **Espressif** Core [**Boards** Manager] **1.0.5**"?  Because after that, when people follow your instructions to "Download latest ZIP file from https://github.com/espressif/esp32-camera.git into the esp32-cam subfolder", they get **Library** (not Boards) Manager content in ver. **1.0.6**.  The only way I can see to get ESP32 ver **1.0.5** of anything is through the Arduino IDE Boards Manager getting **Espressif** as I quoted above.

I also wonder if, or why not do I not need anything **ESP8266**?

I will eventually fix all of this...I always do.  But if you can think about this and give some more details it might save me a month.

Thank you!

arkhipenko, 2021-04-14:
Espressif themselves call it "Arduino core for the ESP32"  - https://github.com/espressif/arduino-esp32
If they call it something else in the Boards manager - that's their inconsistency. 

I just recompiled everything in 1.0.6 and it works - so you are making a mistake somewhere in the process. 
Are you using Arduino IDE? 

1. Pick  the camera sketch you want to implement - there are three options
2. Download the entire repo as a zip file from https://github.com/espressif/esp32-camera.git  into the sketch folder
3. Delete 'examples' folder from the archive
4. unzip using -j option so all files are in the same folder

should work

Why are we talking about **esp8266** ?  That board has nothing to do with this.

kenneth558, 2021-04-14:
Thank you!  That all helps

kenneth558, 2021-04-14:
I think Espressif's new names are different - "arduino-esp32-master.zip" instead of what your instructions refer to as "esp32-cam-master.zip"; and they no longer include the 'examples' folder.  Do you see that, or am I in the wrong place?

arkhipenko, 2021-04-14:
Yep, you are in the wrong place. The zip file is from here: https://github.com/espressif/esp32-camera.git

arkhipenko, 2021-04-14:
Reference to Arduino Core is for configuration of Arduino IDE only - nothing to download

kenneth558, 2021-04-14:
Part of the problem is that Espressif has two very similar named repositories, and I got into their "espressif/arduino-esp32" not using your link into " espressif/esp32-camera".  

**But the other part is that they DID change the name from "esp32-cam-master.zip" like you have to "esp32-camera-master.zip"**

You might want to fix that (and maybe warn to avoid "espressif/arduino-esp32" which is way more current [9 days vs 2 yrs] than your code works with).

Thank you!!!!

closed 2021-04-14 by arkhipenko