arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #14: Compile error after update

krambriw, 2021-06-18:
Hi, compiling your original examples work fine on my debian laptop! All of them compiles and when uploaded to my esp-eye it works. But if I follow the guide to update it doesn't compile anymore:
```
Download latest ZIP file from https://github.com/espressif/esp32-camera.git into the esp32-cam subfolder
Delete examples folder from the archive
unzip using unzip -j esp32-cam-master.zip command. This will place all files in the same folder

```
It complains 
`soc/system_reg.h: No such file or directory
`

Any guidance really appreciated,
Best regards, Walter

fingalo, 2021-07-01:
I also got this issue. My solution was as follows:
1. delete all files in the arduino project except .ino and camera_pins.h
    Not 100% sure if deletion is needed as all files will be overwritten by new ones.
2. unpack the espressif git to the directory as instructed (delete examples)
3. ... but some files are overwritten as there are several versions of some files, so from the git source copy the file
 `esp32-camera-master\esp32-camera-master\target\esp32\ll_cam.c`
4. delete the files coming from the test directory

If I remember correctly this is all.
I just compiled and identified files with compile errors and removed them.

regards
fingalo

arkhipenko, 2021-07-01:
Updated the README - should work (for now).

closed 2021-07-01 by arkhipenko