arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #16: Procedure seems a bit confusing

Gun-neR, 2021-07-21:
Hello.  I am not new to Arduino, but even with years of experience, I am not fully understanding your directions.

1.  You don't mention it, however, I assumed you meant to first download/extract your zip, and THEN download the espressif zip _into_ the `esp32-cam` subfolder _from your_ extracted zip.  But what if I want to use one of the other two options: `esp32-cam-rtos` or `esp32-cam-allframes`?

2.  Done

3. Delete some files from "the sketch" folder"??  You mean _your_ esp32-cam folder??? If so, why not just not have them there in the first place?  That is what I did, however this step just seems to cause compiling issues later.

4.  Done

5.  Again, unzip into what folder??  Your _esp32-cam_ one?  That is what I did.

Phantom step 6. After making the boards type selection and new WiFi config file, I tried to compile, and of course it wants one of the files I just deleted..

`ov2640.h: No such file or directory`

Even after redownloading, retrying after a few times, I am clearly misunderstanding your intended folder structure.  Can you elaborate please?

Gun-neR, 2021-07-21:
Well, I got it working...

Step 1 - Installed in the folder of feature choice (`esp32-cam` `esp32-cam-rtos` or` esp32-cam-allframes`) 

Step 3 - I just skipped it 😉   If not deleting some files causes issues, I can't tell yet.

Step 5.  again, I just used the folder of feature choice (`esp32-cam` `esp32-cam-rtos` or` esp32-cam-allframes`) 

But, I also had to add in the horizontal mirror option for my camera as you only had the vertical flip option.
```
  sensor_t* s = esp_camera_sensor_get();
  s->set_vflip(s, true);
  s->set_hmirror(s, true);
```

arkhipenko, 2021-07-21:
Seems like you have figured it out.

closed 2021-07-21 by arkhipenko

Gun-neR, 2021-07-21:
I still don't understand what files you said to delete, nor why.  

And someone less experienced with "figuring it out" would probably still be stuck, so I still recommend a bit more clarity in the steps.

Other then that, it is working good for me and a nice upgrade for these little cameras 👍

arkhipenko, 2021-07-21:
The OP assumes you already have the git repo cloned on your machine somewhere.
In those 3 folders (1 per different frame serving method) you have my code and a lot of espressif files.
You need to download a zip from the espressif repo ONLY if you want to update the drivers. But you do not need to - everything works and compiles as-is. 
Now, if you **do** decide to refresh the files, that's when you need to delete almost all the files (all **except** `esp32-cam*.ino` and `camera_pins.h`).
By doing that you are deleting all OLD espressif files from the sketch folder, because you are going to replace them with a FRESH ones from the espressif repo's ZIP file. 
Makes a little more sense now?

Gun-neR, 2021-07-21:
Thanks, the clarity of the OPTIONAL espressif repo steps is nice.  As it the reasoning.

Now, after your explanation and manually looking/comparing through all the files/folders, I can better understand the unzip method you describe... as in you need to strip all the _espressif repo_ files **out** of their respective subfolders and into the root folder of the chosen frame serving method.  That wasn't clear to me as you originally have the entire `esp32-camera-master.zip` in the same root folder, so I thought the new one had to replace it (just unzipped into its folder this time).

Anyhow, in my experience many end users and developers generally see things differently, the latter tending to assume the formers thought process would understand those dev nuances.  Thankfully I have progressed far enough from end user territory to (sometimes) see both sides, so I tend to explain things in crayon, and hope others do the same for the poor noobs, as I once was 😄

Gun-neR, 2021-07-21:
PS, Thank You!! for even making this... It solved many issues I had with using these wonderful little camera boards.  Now to get a hold of one of the newer ESP-EYE boards.

arkhipenko, 2021-07-21:
You are welcome. It was a fun little hack.

arkhipenko, 2021-07-21:
Check this one out - another hack which actually could be interesting to combine with the esp32-cam repos since it adds another 4MB of space to the storage. 

https://github.com/arkhipenko/ESP32Himem

yozik04, 2021-07-25:
@arkhipenko Why you do not commit `platformio.ini` to the repo?

arkhipenko, 2021-07-25:
I don't really use platform.io
Do you have a file I can use?