arkhipenko/esp32-mjpeg-multiclient-espcam-drivers #17: ll_cam.c:384:34: error: 'CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX' undeclared (first use in this function)

ericfont, 2021-07-22:
I believe I followed your instructions correctly.  But I get this when trying to compile:

```sketch/ll_cam.c: In function 'll_cam_calc_rgb_dma':
ll_cam.c:384:34: error: 'CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX' undeclared (first use in this function)
     size_t dma_half_buffer_max = CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX / 2 / cam->dma_bytes_per_item;
                                  ^
sketch/ll_cam.c:384:34: note: each undeclared identifier is reported only once for each function it appears in
exit status 1
'CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX' undeclared (first use in this function)
```

Note, the zip I downloaded was last updated two dates ago (tree is at this commit: https://github.com/espressif/esp32-camera/tree/6a9497bbe909165663d958986e621e98dabcf994)

ericfont, 2021-07-22:
I notice that is defined here:

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/config/sdkconfig.h

Maybe that header isn't included but needs to be included?

ericfont, 2021-07-22:
according to these search results:

https://github.com/espressif/esp32-camera/search?q=CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX

that `CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX` is used but not defined in espressif/esp32-camera

ericfont, 2021-07-22:
it compiles if I copy

`#define CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX 32768`

and paste it into the top of ll_cam.c

Maybe some of the driver code has changed.

ericfont, 2021-07-22:
and then it successfully compiles and I can flash to my ESP32-CAM and I can access the stream in my browser!

ericfont, 2021-07-22:
as an aside, this is how my test came out: https://www.youtube.com/watch?v=6Md69RYzKV8

(I have to admit I'm a little disappointed that it feels like a whole second from when a visual change in the camera propagates to my end browser.  I guess that is a result of how buffering is handled.)

arkhipenko, 2021-07-22:
Hi there. I hope you understand that this whole "procedure" that I concocted is just a dirty hack, not a proper way to include a submodule. It will have to potentially change every time Espressif changes something in there code.
As far as I am aware, unfortunately, this is so far the only way to make it work on Arduino with multiple clients, so we will have to continue to find ways to hack through new updates.
Thanks for finding this one. I will looking into it. Which esp32 board did you use? 
Last comment: if the latest drivers do not work, you can always use my repo as-is. It works as of whatever date I last tested it.

ericfont, 2021-07-22:
ESP32-CAM written on front and ESP32-S written on back (purchased from https://www.amazon.com/gp/product/B08MJJTFN9)

My arduino IDE is set wtih:
[image]

ericfont, 2021-07-22:
when you say "make it work on Arduino" do you mean specifically "Arduino IDE"?  Are you suggesting it may be better to just forgo the use of Arudino IDE and instead use esp-idf?  (In that case I'm going to try with a clean fresh git repository but I will point to this and credit your code as an initial commit cause I don't know how to transfer git commits to a fresh repo)

arkhipenko, 2021-07-23:
No, I mean it is meant to work with Arduino Core. There is a lot more to it than just IDE. 
And yes, if you want to use IDF environment, it is completely different, but please go ahead!